mcer12/Flora-ESP8266 #4: Throwing exception at launch

geoffreyheller, 2021-03-09:
Cool project!

I have built the Flora 1.0 (yes I know there's a newer one coming, but I had already ordered the boards and parts earlier).  When I compile and upload the github code I get an exception thrown at launch.

I have narrowed this down to some problem with the NeoPixel strip library.  If I comment out all the strip calls, the wifi starts up and I get the website where I can configure the wifi ssid, password, and display information.

I am wondering if I am building with a different set of libraries than you?  Or perhaps my compile options are different than yours?

Can you please provide a list of the specific library versions you are using?

Here are my settings from the Arduino IDE:
Board: Generic ESP8285 Module
Builtin Led: 2
CPU Frequency: 80 Mhz
Crystal Freq: 26 Mhz
Flash size:  2MB (FS:64KB OTA~992KB)
Reset Method: dtr (aka nodemcu)
IwIP Variant: v2 Lower Memory
VTables: Flash
Exceptions: Legacy
SSL: All SSL ciphers (most compatible)

Thank you!
-G.

mcer12, 2021-03-09:
Hi, first thing I see is that flash size is wrong, it should be 1M + 64SPIFFS for ESP-M2 (unless you designed your own board with different module ;) ) Another thing is that this sketch relies heavily on very fast timing, so definitely set CPU frequency to 160Mhz.
I use neopixel library 2.5.7, I will update to latest but I doubt that to be the culprit.

mcer12, 2021-03-09:
Pixel count is set to 12 because the 6 digit version has 12 leds :) The ESP has no way of knowing how many leds there really are so it doesn't make any difference.

geoffreyheller, 2021-03-09:
Ok I have set my flash size to 1M + 64SPIFFS as per the above and have also set the CPU freq to 160Mhz.

I am compiling with the following libraries:
NeoPixelBus by Makuna version 2.5.7
Time by Michael Margolis version 1.6.0
Ticker by Stefan Staub version 4.1.0
Timezone by Jack Christensen version 1.2.4 (note that this produces a compiler warning as it is targeted for AVR arch.  Possibly want to replace this with Timezone_Generic)

I am still seeing the exception happening.  Here's the exception decoded from the stack trace:

```
Exception 0: Illegal instruction
PC: 0x40209268: SPIClass::transfer(unsigned char) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SPI\SPI.cpp line 308
EXCVADDR: 0x00000000

Decoding stack results
0x40220b0b: __swbuf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/wbuf.c line 82
0x401002cf: TimerHandler() at C:\Users\Geoffrey\Documents\PlatformIO\Flora-ESP8266\Firmware\FLORA_FIRMWARE/VFD_fns.ino line 50
0x40100418: timer1_isr_handler(void*) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 36
0x40100460: timer1_isr_handler(void*) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 43
0x4010135c: umm_malloc(size_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x4020f4fa: String::toUpperCase() at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 811
0x40101411: umm_realloc(void*, size_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 586
0x40100f28: get_poisoned(void*, size_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc/umm_poison.c line 117
0x40101629: umm_poison_realloc_fl(void*, size_t, char const*, int) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc/umm_local.c line 140
0x40100fee: check_poison_neighbors(uint16_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc/umm_local.c line 59
0x401010f9: umm_malloc_core(size_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 458
0x401003e5: __wrap_spi_flash_read(uint32_t, uint32_t*, size_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_phy.cpp line 309
0x40100f28: get_poisoned(void*, size_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc/umm_poison.c line 117
0x402098bf: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 432
0x4020e549: String::invalidate() at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 140
0x40209e29: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 468
0x402068a3: startConfigPortal() at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/WString.h line 230
0x40211875: uart_write(uart_t*, char const*, size_t) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 509
0x4020cf4c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x4020cf4c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x4020d694: Print::println() at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Print.cpp line 186
0x40208670: setup() at C:\Users\Geoffrey\Documents\PlatformIO\Flora-ESP8266\Firmware\FLORA_FIRMWARE/FLORA_FIRMWARE.ino line 267
0x4020f904: loop_wrapper() at C:\Users\Geoffrey\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 194
```
Any ideas what this might be?

mcer12, 2021-03-09:
Oh you're correct and sorry I forgot about this... I had to edit the SPI library to support ISR timer. Pull the latest github repository and copy the library to arduino libraries folder. Reopen the sketch and flash it, should work now.

It will give you duplicate SPI library warning but it should correctly take the edited one.

geoffreyheller, 2021-03-10:
The SPI library update seems to have fixed the issue.  Thanks for following up!

mcer12, 2021-03-10:
Great to hear :)

closed 2021-03-10 by mcer12