mcer12/Flora-ESP8266 #5: PCB assembly question

geoffreyheller, 2021-03-10:
Would it be possible to include some photos of the assembled PCB in the resources directory?

So for example, I'm not certain of the orientation of the VFD modules (top vs bottom) with respect to the PCB.   It would be very helpful to see a few photos of a completed and working assembly before I go ahead and solder the VFDs in.

Thank you!

mcer12, 2021-03-10:
I've added pcb images, should be clear now :) Also added some notes for tesing voltages.
**IMPORTANT:**
Double and triple check if the headers and usb connector are orented properly! I've made this mistake several times, especially soldering the usb connector backwards :D

closed 2021-03-10 by geoffreyheller

mcer12, 2021-03-11:
Check out this:
https://github.com/mcer12/Flora-ESP8266/wiki/IV22-Assembly-notes

geoffreyheller, 2021-03-11:
Wow!   Awesome notes.

I did not expect the glass to be so misaligned from the digits.  That's
great to know before I start soldering.
-G.

geoffreyheller, 2021-03-11:
I'm still unclear as to which side of the board is the top and which is the
bottom with respect to the IV22 digit orientation.

It would be great if there was an additional photo showing the tubes
attached to the board in such a way that I can see the board orientation
and the decimal points.

Best Regards,
-G.

mcer12, 2021-03-11:
That’s easy! Short headers = down, long headers = up :)