mcer12/Flora-ESP8266 #10: Wifi DHCP connectivity issue

geoffreyheller, 2021-03-19:
I'm having trouble getting reliable startup of my Flora build.

It frequently fails to complete DHCP initialization and drops into config mode.   I am using an ASUS RX-88u Wifi router.

Serial debug output looks like this:
```
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 24
cnt 
state: 5 -> 2 (2c0)
rm 0
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 9
cnt 

connected with Foonet, channel 2
dhcp client start...
pm open,type:2 0
WIFI: Failed to connect to: Foonet, going into config mode.
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
mode : softAP(ca:2b:96:70:9b:d8)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
IP address: 10.10.10.1
```
From the above it appears that it is connecting to my Foonet Wifi, but then dhcp fails to resolve and it drops back to config mode.

I'm going to look into this a bit more on my end to see if I can come up with a solution.

mcer12, 2021-03-19:
This could be anything from power issues to some misunderstanding between ESP and the router. First thing I would try is to dig in to the router settings. Maybe there's something timing out.

mcer12, 2021-03-19:
Also make sure that the signal is good (close to the router), maybe it's that...

geoffreyheller, 2021-03-19:
I will experiment with my router settings and see if I can cause an improvement.
That said, I have several ESP8266 here as part of ESPHome and I don't have an issue with them connecting. I'm not sure what the difference is.

mcer12, 2021-03-19:
From one forum:
_So the answer was that the chip does not support WPA TKIP. It acts like it will work but it just does not get an IP address. The failure is not very graceful. So the system thinks it is connected when it is not._

geoffreyheller, 2021-03-19:
I was able to resolve this issue by disabling ASUS's "Smart Connect" feature.  

See related ASUS support article: https://www.asus.com/support/FAQ/1042475/

closed 2021-03-19 by geoffreyheller