mcer12/Flora-ESP8266 #8: Daylight savings time incorrect

geoffreyheller, 2021-03-19:
I live in a PDT time zone.  When I enable the Daylight Savings Time checkbox in the config menu my clock reads four hours different from our local time.

I suspect this is the combination:
1) Our normal PST offset is -8h (-480 in the config menu), but we are in PDT so there's a missing hour
2) The TimeChangeRule is hard-coded in FLORA_FIRMWARE to EDT:
```
TimeChangeRule EDT = {"EDT", Last, Sun, Mar, 1, 120};  //UTC + 2 hours
TimeChangeRule EST = {"EST", Last, Sun, Oct, 1, 60};  //UTC + 1 hours
Timezone TZ(EDT, EST);
```
I suspect that this value is either not getting saved to JSON from the config menu or is not getting loaded properly.

mcer12, 2021-03-19:
These are just default values, they are ignored. You have to set the timezone in Config portal and dont forget that it's in minutes, not hours ;)

mcer12, 2021-03-19:
Please double check the settings as there might be an error. And if it still doesn't seem to work correctly, please send me a screenshot of your config portal.

geoffreyheller, 2021-03-19:
Here are my timezone and dst settings.   At 11:05am PDT local time I am seeing 7:05 reading out on the clock display.

[image]

mcer12, 2021-03-19:
Oh it's UTC offset, not relative offset :) So it should be -420, not 60 in this case.

geoffreyheller, 2021-03-19:
Aha!  Yes that did fix it.  Thank you!

closed 2021-03-19 by geoffreyheller