mcer12/Flora-ESP8266 #9: Feature Request: LED brightness and pattern control

geoffreyheller, 2021-03-19:
For my v1.0 build I'm getting the entire 5-LED center column pulsing brightly once per second.  

It would be nice if I could easily reduce the brightness to match the VFD output.  

Also, it would be nice if we could optionally select an LED display pattern that looks more like a colon.   So perhaps three display modes such as:
1) Full column (default)
2) Top two LEDs and bottom two LEDs
3) Second and Fourth LEDs only (odd LEDs)

I'm happy to look into doing this programming if you like.

mcer12, 2021-03-19:
You will likely have to modify the color balance to your version of WS2812. You will have to tinker with the brightness and color balance with the colon plate attached. Feel free to reprogram the colon but since the light will be guided through the colon anyway, I don't see any reason for this ;)

mcer12, 2021-03-19:
You can also turn the colon off in the config portal if it's annoying

geoffreyheller, 2021-03-20:
Ok now that I have 3d printed the case I understand that the colons are a sort of light pipe.  I'm going to try printing them in clear on my SLA printer.

I still think it might be nice to be able to easily customize the color of the colons.  As you say, the color balance might vary between different WS2812's and need adjustment.

closed 2021-03-20 by geoffreyheller

mcer12, 2021-03-21:
Yes, I thought of that but since you have to set the color for each brightness setting separately, it would get kinda bloated... 

I think it's actually a bit better if you print it with FDM (I used clear PLA), it scatters the light better even when the top is not sanded. If you sand the top, it looks perfect. SLA might actually be too clear.