mcer12/Flora-ESP8266 #21: 12 hour format

nikkisstuff, 2021-12-04:
The 12 hour format is showing 00 midnight instead of 12. as I understand it 00 is 24 hour format.

mcer12, 2021-12-04:
@nikkisstuff What version of firmware are you running? I just tried it and seems to work fine. You can see FW version in config portal (at the bottom).

mcer12, 2021-12-04:
Sorry, gotcha... I'm not used to 12h format on digital clock so I tried it, saw 0:00 at midnight and my brain shorted thinking it was correct :D Will fix in a moment

mcer12, 2021-12-04:
Made a quick update, so you can flash the binary via OTA:
https://github.com/mcer12/Flora-ESP8266/releases/tag/5.1.2
https://github.com/mcer12/Flora-ESP8266/wiki/OTA-update

nikkisstuff, 2021-12-04:
perfect thanks You.

closed 2021-12-04 by nikkisstuff