mcer12/Flora-ESP8266 #25: PCB/Schematic files

harrehlah, 2024-04-04:
Hey mate,

I've been learning and playing around with kicad on my own stuff and now I am wanting to modify your design to take a daughter board to support 3x more IV3s for a temperature display in a new enclosure I have designed - am I able to get a copy of the PCB files? I'm pretty far away skill wise from being able to even replicate your design just yet.

Cheers!

technotrasher, 2024-05-13:
unfortunately the author did not provide pcb Flora-ESP8266
you will have to create the board yourself. the author has errors. It also needs improvement - to connect a time module.
if you have a pcb, could you share?

mcer12, 2024-05-13:
@technotrasher Can you elaborate on the errors?

technotrasher, 2024-05-13:
needs improvement. rare AP7361C-33E-13 - replacement for the common LD1117S33TR. ESP M-2 is a conductor between EN - 10uF - GND. 3.3V is not correct!
in config_portal.ino replace bool startServer() - void startServer(), in FLORA_FIRMWARE.ino remove the unnecessary #include <TimeLib.h>.
NeoPixelBus library is not older than 2.6.6!!!

technotrasher, 2024-05-13:
I also want to add a time module. We have frequent power outages. The watch works fine from the power bank. but there is no Internet access.

harrehlah, 2024-05-14:
> @harrehlah I will add easyeda source for IV3 momentarily, just be aware that you will have to redesign the power supply if you want to drive more tubes. It will probably require USB PD, you won't be able to drive that many tubes from simple 5V USB.

Thanks mate! I appreciate it!

technotrasher, 2024-05-14:
NeoPixelBus by Makuna v. 2.8 - "D:\Flora IV-6\FLORA_FIRMWARE\neopixel_fns.ino: In function 'void updateColonColor(RgbColor)':
D:\Flora IV-6\FLORA_FIRMWARE\neopixel_fns.ino:79:67: error: call of overloaded 'LinearBlend(RgbColor&, RgbColor, double)' is ambiguous
   79 |     colorMed = RgbColor::LinearBlend(color, RgbColor(0, 0, 0), 0.1);
      |                                                                   ^
In file included from c:\Users\technotrasher\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoColors.h:34,
                 from c:\Users\technotrasher\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:38,
                 from D:\Flora IV-6\FLORA_FIRMWARE\FLORA_FIRMWARE.ino:25:"