mcer12/Flora-ESP8266 #26: access point

guus46a, 2024-11-28:
Hello, i have esp12 and wemos programmed, but i get no access point.
Esp12 gives
14:48:07.860 -> load 0x4010f000, len 3584, room 16 
14:48:07.860 -> tail 0
14:48:07.860 -> chksum 0xb0
14:48:07.860 -> csum 0xb0
14:48:07.860 -> v2843a5ac
14:48:07.860 -> ~ld
14:48:07.860 -> 
14:48:07.860 -> [CONF] Failed to mount file system
14:48:07.860 -> [CONF] Failed to read config file... first run?
14:48:07.860 -> [CONF] Creating new file...
14:48:07.860 -> [CONF] Failed to open config file for writing
14:48:07.860 -> [CONF] No0, len 3584, room 16 

in monitor, and wemos givves
14:52:21.959 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
14:52:21.959 -> 
14:52:21.959 -> load 0x4010f000, len 3584, room 16 
14:52:21.959 -> tail 0
14:52:21.959 -> chksum 0xb0
14:52:21.959 -> csum 0xb0
14:52:21.959 -> v2843a5ac
14:52:21.959 -> ~ld
14:52:22.000 -> 
14:52:22.023 -> [CONF] No credentials set, going to config mode.

please help

mcer12, 2024-11-28:
Make sure you select to completely erase flash and also to use version 2.6.x of the esp8266 boards in arduino ide.

guus46a, 2024-11-28:
I have 2.7.4, and try to find 2.6.x but 2.7.4 can not go lower

mcer12, 2024-11-28:
Yes it was 2.7.x sorry :D 2.7.4 is a perfect version to use. Did you follow wiki?
https://github.com/mcer12/Flora-ESP8266/wiki/Firmware-and-flashing

guus46a, 2024-11-28:
I am trying  to change a working clock to get esp on it. That is difficult because it has ds3231, stc programmed ic an an ic i do not know what it is. As i use your bin than i get homepage and can set credentials but after that i can not find page anymore. I have removed the stc ic and found data, sh-cp and st-cp and thought that it should work, but i think there is more going on because vfd's show 888
[image]

mcer12, 2024-11-28:
Once you connect to wifi, you can only access the config portal via local IP address as mentioned on wiki.
As for trying to botch it onto the other clock, you need to understand the circuit and probably remove the original IC on the board. Just using ESP8266 and my firmware won't work, it will take quite alot of time nad I can't help you with that.

guus46a, 2024-11-28:
I have removed the ic on board and tubes are working with 74hc595n and uln2003. Seconds tube ic has 14 free for data when ic is out and also pin 11 an 12 are connected from all 595's and free when ic is out. Website is now working but tubes still light up as 888.

technotrasher, 2024-12-05:
> I am trying to change a working clock to get esp on it. That is difficult because it has ds3231, stc programmed ic an an ic i do not know what it is. As i use your bin than i get homepage and can [...]

I'm working on adding a ds3231 module to my watch. the board is already ready. Now all that remains is to rewrite the program itself. I did this for the watch on NIXIE. and I have ready-made solutions.

guus46a, 2024-12-05:
The only way i could solve my problen is to put an esp sync to ds in the esp and when i want to sync the clock, i put 3.3 volts on esp and clock givves right time.

mcer12, 2025-03-24:
Closing this issue as the further discussion is not really related to this project.

closed 2025-03-24 by mcer12